tbfleming/jscut #34: Establishing an XY origin

wcpenney, 2015-01-07:
First of all, thanks for all of the hard work on this. jscut really is an awesome app.

The first few times I created an SVG file (with Inkscape) and generated gcode with jscut, it didn't make much sense to me. Y coordinates originate in the upper left, and use negative coordinates while Inkscape appears to use the lower left and positive coordinates. After applying some google-fu, I found out about Inkscape's wonky coordinates/ruler system.

The "set origin to lower left" button doesn't appear to work until after I generate a toolpath. Often the 3D view re-sizes on(during?) regeneration giving the appearance that nothing happened until I export and read the gcode or resize the browser.

 In addition, setting the origin to the lower left chooses the leftmost point of a selected object. In the case of a circle, I may not be able to accurately find that point to square my stock in relation to it before milling.

I have a "template" file setup in Inkscape with the page set to the dimensions of my Shapeoko. Since the page size is obvious in Inkscape and tied to X and Y, would it be possible to allow selecting a page corner as the origin? I think asking a user would point out the difference between what Inkscape presents and what it saves in the SVG file, eliminate a lot of initial head scratching, and eliminate a lot of questions on the forums/G+/here, etc.

Working with one positive axis and one negative axis sounds like it will get really confusing while configuring Grbl come homing/limit switch time. Would it be possible to have a selection to convert negative coordinates to positive ones (or an option to just set all coordinates positive or all coordinates negative)?

As a final suggestion, maybe roll all of the XYZ coordinate stuff into a single box and make it an early step?

Thanks for reading this. I hope it makes some sense and I'm not missing something obvious.

tiger12506, 2016-01-05:
Just recently ran in to the exact same problem. I would like to second that this is a good idea, to allow setting a page corner as the origin.

I've been trying to workaround by creating dummy paths that coincide with the edges of my stock, but then this has to be offset by the width of the cutter in order to be truly accurate, and it's all just too confusing.

neocogent, 2019-03-14:
This issue drives me batty too. Plus when clicking "Zero Lower Left" it aligns to tool path rather than page co-ords or even actual cut outer edge - so you have to manually figure out how much to offset to get back so work piece location matches up. The whole point of this is so we don't have to calculate things manually so this makes it so much more prone to error.

tbfleming, 2019-03-14:
I don't enhance jscut anymore. LaserWeb4 is its successor.

Patrick762, 2023-02-02:
@tbfleming could you please add this info to the readme of this project?

tbfleming, 2023-02-02:
Updated readme