tbfleming/jscut #35: Saving gcode to chilipeppr doesn't work?

matthijskooijman, 2015-01-20:
I just tried saving the generated gcode to chilipeppr.com, but it says "Not logged into http://chilipeppr.com/" (in an easy-to-miss popup in the background). I just logged into chilipeppr.com (and refreshed both chilipeppr and jscut afterwards), so that cannot be true.

Any thoughts?

tbfleming, 2015-01-20:
Does the upper-right corner of chilipeppr give your user name or say "login"?

[image]

matthijskooijman, 2015-01-20:
It shows my gmail address, yes.

tbfleming, 2015-01-20:
Odd. I'll ask John if he has any ideas.

matthijskooijman, 2015-01-20:
If there's anything else I should try, let me know. If you can give a quick summary of how the login code works (and/or where to find it), I could have a closer look myself.

tbfleming, 2015-01-20:
The login check happens at https://github.com/tbfleming/jscut/blob/gh-pages/js/jscut.js#L779

matthijskooijman, 2015-01-20:
Hm, I guess chilipeppr is not returning any user. If I try it in the console:

	$.getJSON("http://www.chilipeppr.com/datalogin?callback=?").done(function(content){console.log(content)})
	Object {readyState: 1, getResponseHeader: function, getAllResponseHeaders: function, setRequestHeader: function, overrideMimeType: function…}
	VM240:2 Object {CurrentUser: null, LoginUrl: "https://www.google.com/accounts/ServiceLogin?servi…Sjqyegzner1iPArhYoATIU2mTvwfLsh8zEhMgSFYKFXDuLRyc", LogoutUrl: "http://www.chilipeppr.com/_ah/logout?continue=http…nue%253Dhttp://www.chilipeppr.com/%26service%3Dah"}

tbfleming, 2015-01-20:
The odd part: it works for me

johnlauer, 2015-01-20:
See if you are signed into www.chilipeppr.com and chilipeppr.com. Google seems to treat them differently.

matthijskooijman, 2015-01-20:
I'm shown as logged in on both. I tried logging out on www.chilipeppr.com and logging in again, but that didn't fix it (not sure if it changed anything).

Now, when I go to http://www.chilipeppr.com/datalogin directly, it shows my login. On http://chilipeppr.com/datalogin it shows null.

When I open chilipeppr itself, it makes a request to http://www.chilipeppr.com/datalogin?callback=jQuery21007010580077767372_1421782156152&_=1421782156153, with an `ACSID` cookie (and some Analytics cookie). When I open jscut and try saving to chilipeppr, it makes a request to http://www.chilipeppr.com/datalogin?callback=jQuery21102604281560052186_1421782000018&_=1421782000019 but does _not_ send any cookies (for reasons I don't really understand). It seems this lack of a cookie is probably the problem here.

I should have mentioned this before, but I'm using Chromium 39 for testing. I just also tried Google Chrome 39, which works as expected (and sends the cookie). So perhaps Chromium is somehow more pedantic about cross-domain requests here?

tbfleming, 2015-01-20:
Ah-ha! I wonder if that means a future version of Chrome will break this.

johnlauer, 2015-01-20:
Interesting. This is a jsonp trick. Jsonp is pretty standard now but I could see tighter security in future. There are other ways to pull this off

tbfleming, 2015-01-20:
Another data point: I just did a fresh install of Ubuntu 14.04 on a VM then installed Chromium 39. It worked. Here's the version string:

Version 39.0.2171.65 Ubuntu 14.04 (64-bit)

Which distribution are you using?

matthijskooijman, 2015-01-20:
You left out the most important part of the datapoint: Does it work? :-)

I have Version 39.0.2171.71 Built on 8.0, running on Debian 8.0 (64-bit)
(And I'm actually running Debian testing too)

tbfleming, 2015-01-20:
Yes.

tbfleming, 2015-01-20:
Wow Debian has slow mirrors. Ubuntu spoiled me. Another data point soon.

tbfleming, 2015-01-20:
Slow install after fetch also. I wonder how Ubuntu sped it up.

tbfleming, 2015-01-20:
This failed:
Debian Jessie Beta 2
Version 39.0.2171.71 Built on 8.0, running on Debian 8.0 (64-bit)

tbfleming, 2015-01-20:
If we're lucky, then .71 has a bug that didn't exist in .65 and was fixed by .99.  If we're unlucky, then Chromium tightened up security and it will eventually make it to Chrome.

matthijskooijman, 2015-03-26:
I just upgraded chromium and it still doesn't work there: Version 41.0.2272.76 Built on 8.0, running on Debian 8.0 (64-bit)

Upgraded google-chrome does work: Version 41.0.2272.101 (64-bit)

Note that "work" means that no login error is shown and /datalogin returns correct data, but instead "undefined" is shown in the upper right of jscut and I can't see any file in chilipeppr. No time to dig into this right now, but at least it seems to be an unrelated issue.

So, it seems that this is not just a bug in a single Chromium build - Debian is building Chromium with some change, patch or configuration that breaks these requests for some reason.