tbfleming/jscut #36: Allow non-closed lines for the engrave option

carlosgs, 2015-03-04:
Hi, engraving non closed lines can be a very useful function to have. Right now it is a bit annoying as every design needs to have closed shapes.

[image]
[image]
[image]

I haven't yet found an option to do this easily, what are your thoughts?
By the way I'm loving Jscut, thank you so much for this amazing tool

tbfleming, 2015-03-04:
It's a complicated problem:

1. I use the browser's built-in hit testing for selecting objects. That doesn't work well with open paths; they're hard to click on.
2. jsclipper mangles open paths; I'd have to bypass all code which uses it when open paths are present.
3. Most of jscut's code base assumes closed paths; I'd have to bypass much of the existing code when open paths are present.

jscut automatically closes an open path if a user somehow manages to click on it; this prevents jsclipper and the rest of jscut from mangling things.

carlosgs, 2015-03-05:
Oh, thanks for the explanation, that makes total sense. What a pity then!

I have the crazy thought of changing the closure behavior, instead of just joining start-to-end, duplicate the path and join the extremes. But yeah that'd imply a polygon with zero area, which wouldn't be selectable at all... etc etc

Anyway no worries as it is very possible to live without support for open paths. Thanks again for your quick answer!!

closed 2021-04-15 by CarlosGS