tbfleming/jscut #37: Cannot select objects in view to create paths

Stephen337, 2015-03-06:
When trying to select objects the click to select the object has to be super accurate on the line or the object cant be selected. To select the objects the click has to be right on the line and on my CNC machine the screen is low resolution and I can t get the accuracy needed.

Would it be possible to make the line selection easier when there are simple shapes. For example try loading and selecting the shapes in the svg file:

<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   version="1.1"
   width="170mm"
   height="120mm"
   viewBox=" 0 0 602.3627 425.1972"
   id="svg2"
   inkscape:version="0.91 r13725"
   sodipodi:docname="caseplans.svg">
  <metadata
     id="metadata32">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <defs
     id="defs30" />
  <sodipodi:namedview
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1"
     objecttolerance="10"
     gridtolerance="10"
     guidetolerance="10"
     inkscape:pageopacity="0"
     inkscape:pageshadow="2"
     inkscape:window-width="1600"
     inkscape:window-height="838"
     id="namedview28"
     showgrid="false"
     inkscape:zoom="1.3664076"
     inkscape:cx="242.21269"
     inkscape:cy="228.20495"
     inkscape:window-x="-8"
     inkscape:window-y="-8"
     inkscape:window-maximized="1"
     inkscape:current-layer="svg2" />
  
  <g
     transform="translate(15.673241,21.528013)"
     id="g4">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 0,0 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 L 0,35.4331 Z"
       id="polygon6"
       inkscape:connector-curvature="0" />
  </g>
  
  <g
     transform="translate(222.11261,20.06432)"
     id="g8">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 L 0,35.4331 0,0 Z"
       id="polygon10"
       inkscape:connector-curvature="0" />
  </g>
  
  <g
     transform="translate(413.91504,19.332473)"
     id="g12">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 18.425212,0 17.007888,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 17.00789,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 -17.00789,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -17.007888,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 z"
       id="polygon14"
       inkscape:connector-curvature="0" />
  </g>
  
  <g
     transform="translate(410.98765,227.96738)"
     id="g16">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 18.425212,0 17.007888,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 17.00789,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 -17.00789,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -17.007888,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 z"
       id="polygon18"
       inkscape:connector-curvature="0" />
  </g>
  
  <g
     transform="translate(17.136933,227.96738)"
     id="g20">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 18.425212,18.425212 17.007888,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 17.00789,0 0,17.007888 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,17.00789 -17.00789,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -17.007888,0 0,-17.00789 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 z"
       id="polygon22"
       inkscape:connector-curvature="0" />
  </g>
  
  <g
     transform="translate(216.25783,227.96738)"
     id="g24">
    <path
       style="fill:none;stroke:#000000;stroke-width:0.90000075"
       d="m 18.425212,18.425212 17.007888,0 0,-18.425212 35.4331,0 0,18.425212 35.4331,0 0,-18.425212 35.4331,0 0,18.425212 17.00789,0 0,17.007888 18.42521,0 0,35.4331 -18.42521,0 0,35.4331 18.42521,0 0,35.4331 -18.42521,0 0,17.00789 -17.00789,0 0,18.42521 -35.4331,0 0,-18.42521 -35.4331,0 0,18.42521 -35.4331,0 0,-18.42521 -17.007888,0 0,-17.00789 -18.425212,0 0,-35.4331 18.425212,0 0,-35.4331 -18.425212,0 0,-35.4331 18.425212,0 z"
       id="polygon26"
       inkscape:connector-curvature="0" />
  </g>
</svg>

tbfleming, 2015-03-06:
Give the objects a solid fill color; that makes them easy to select. I use the browser's built-in hit-testing, which only works well with solid fill.

Stephen337, 2015-03-07:
Thanks, much better!

closed 2015-03-07 by Stephen337