tbfleming/jscut #38: Make all inches/mm

tiger12506, 2015-05-29:
[image]

From my perspective, it looks like the "Make all mm" and "Make all inch" links should convert, but the drop-down should not convert, just read the current values as the specified units.

This is currently not the case, as both the drop-down and the link convert the values in the fields.

g19fanatic, 2015-06-03:
I believe having two ways of changing is unnecessarily redundant. I would opt for changing the drop-down to a label to declare what unit is being used and just keep the "Make all" actions.

tbfleming, 2015-06-03:
There are multiple independent units fields. The "Make all" links set all of them at once.

tiger12506, 2015-06-05:
Ok, so this doesn't actually duplicate functionality then, because the Make all links affect more than just the Material grouping. Got it. I'd still be interested in a "change how we interpret the numbers without converting the numbers" feature, but certainly not a deal-breaker in any way, of course.

I greatly appreciate the work you are doing here.

closed 2016-05-16 by tbfleming