tbfleming/jscut #41: Jscut generating 2 identical cut paths on lowest layer

andrewhodel, 2015-06-05:
##Steps to reproduce issue

* open jscut
* Open SVG -> Local
* Select tocut.svg
* Click "Make all mm"
* Change Thickness to 10
* Change Clearance to 3
* Change Tool Diameter to 6.35
* Change Tool Pass Depth to 4
* Change Tool Cut Speed to 600
* Select Both SVG Drawings (they turn from yellow to blue when selected)
* Click Create Operation
* Change Operation to Outside instead of Pocket
* Change from 4 to 12 Deep for Operation
* Click Generate on Operation
* Look at Gcode, on the lowest layer it runs the same code twice for each object

##Files

tocut.svg - http://pastebin.ca/3018586

tocut.gcode - http://pastebin.ca/3018587 (line 780 it goes to Z-12 and cuts, then on line 1157 it again repeats the same process)

This results, especially when cutting sheets, in the object being cut free from the sheet and then the tool making another pass on the same path at the same depth.  Unless you are standing there holding the cut out part in place it will ruin the cut.

tbfleming, 2015-06-09:
Please use "Save Settings" -> Gist to make reproduction easier.

andrewhodel, 2015-06-09:
https://gist.github.com/anonymous/44f0874b86bdbb47519c

tbfleming, 2015-06-09:
It's a rounding bug; it creates a path at Z=-11.99999... then at Z=-12. The rounding code generates -12.0000 for both values. Workaround for this case: change Pass Depth to 4.01.

andrewhodel, 2015-06-09:
Can you give me the point in the source where this is calculated?

tbfleming, 2015-06-09:
https://github.com/tbfleming/jscut/blob/gh-pages/js/Cam.js#L409
https://github.com/tbfleming/jscut/blob/gh-pages/js/Cam.js#L463
https://github.com/tbfleming/jscut/blob/gh-pages/js/Cam.js#L474

closed 2016-05-16 by tbfleming