tbfleming/jscut #43: Setting Where The Gcode Cuts First.

Akfreak, 2015-06-11:
I am using JSCut to setup OX CNC gantry plates. The parts are beautiful, and everything is almost perfect. However the fist cut on all the parts is the outside boarder of the part. How can I set it so it does all the holes and pocket cuts inside the part first, the finishes with I the outside?   

If the outside is cut first, the clamping is a mute point at the part is no loose. Please tell me how to fix this problem.   PS, I love JS cut. Great job!

tbfleming, 2015-06-11:
Separate the holes from the outside shape (Inkscape's break apart command). In jscut, select the holes first, create a pocket operation, then select the outside and create an outside operation.

tbfleming, 2015-06-11:
[image]

Akfreak, 2015-06-12:
I use illustrator to make all my parts So I guess Ill make the plate one part, then all the holes separate parts. 

What I did was use the Pathfinder tool in Illustrator to exclude the holes from the plate and it changes into as a single part with holes in it.   

I think I know what to do now.. Thanks for your time.  BTW, I started a JScut G+ group today to talk about JScut projects ans such.  I linked all the known resources of JScut. Thanks for making this awesome and easy to use CAM tool. 

Thomas Shue.

closed 2016-05-16 by tbfleming