tbfleming/jscut #44: DXF Support

openhardwarecoza, 2015-06-15:
Now, I have raised this on Google Plus with @tbfleming, I have emailed, I see other people have emailed, and I do have sympathy for the author in saying DXFs are a mess.  

I do agree it is, but I also do believe that whats standing in the way op opensource CNC the most, is the belief that we can adapt the world to us, instead of the other way around.  For many of us to suppport a project, we need to use a project to make money.  Reprap got it rights years ago, it creates cashflow for the developers.   

This here is the same, we have a army of open source machine owners who can make money out of their garage and in turn plow some it back to the author or other projects

BUT, and thats a big one, most of your 'customers' have Never used Inkcape.  CAD has for years been dominated by DXF as the lowest common denominator between CAD/CAD and CAD/CAM.  If I want a part made, I send a DXF to the factory and pickup tomorrow. Simple - no questions about dimension or pixels per unit or none of those crap.

Blaming DXF files with disconnected entities is to me a easy way out - and I am not happy with that. We have developers with skills.... The reprap slicer guys have long figured out how to repair STLs on the fly - and thats in 3D, with DXF we have on less dimension so it cant be that hard!  We have talented developers like @andrewhodel etc who also has a grasp of this....

Heres what I am thinking:

a) We are looking at the DXF at too high a resolution - most commercial packages just assume if the endpoints are withing 0.1mm of each other they are connected?

b)  Adding pre process instructions like "Save as R14 in QCad before opening in JSCut" is completely acceptable

c)  TALK about it. Dont just wipe it under the rug as Impossible! We know its not!

anthonywebb, 2015-06-15:
I hope to be able to contribute to this one.  There are several libraries out there which are capable to parsing through these DXF's and making them a little more usable.  @openhardwarecoza do you have a sample DXF I could play around with?

Here are a couple of the libraries I have in mind:
https://github.com/bjnortier/dxf
https://github.com/johannesboyne/node-dxf
https://www.npmjs.com/package/dxf-parser
https://www.npmjs.com/package/dxf-parsing

I would imagine those should provide enough of a starting support to where we could at least get some sort of limited DXF support rolled it.  Of course I say this without having looked at the JSCUT code at all, and perhaps it is the optimist in me, but definitely seems doable.

tbfleming, 2015-06-15:
Writing parsers is trivial. I do it professionally for stuff insanely more complicated than DXF. The above links do the easy part and leave out the hard part: repair.

I was there in the early years of RepRap when most people submitted broken STL files to Thingiverse and the repair tools were mostly broken and unusable. The forums were flooded with irate people wondering why their STL files didn't print correctly. It took 3-4 years for automatic repair to work half-way decent, but even now automatic repair has unpredictable results for new users.  The ill will from incorrectly handling DXF would be much worse than leaving DXF support out.

Rounding endpoints together isn't enough; figuring out where the closed loops are when segments overlap or are missing is much harder because their are multiple solutions, but only 1 solution will produce what the person who drew it wants. **Choosing to close the wrong loops will produce incorrect cutter movement**.

If someone is willing to write a repair tool that works well, and is willing to relieve me of having to answer support questions for DXF files that have problems after conversion, then I'll be willing to integrate it. Just be aware that #22 prevents incorporating existing GPL code.

anthonywebb, 2015-06-15:
@tbfleming any idea how frequently DXF files need to be repaired?  It seems like generating a valid DXF to begin with would be a priority for the CAD packages out there, and I would assume they do at least a decent job at it?

misan, 2015-06-15:
@anthonywebb It is my understanding that many programs that export to DXF are designed to produce a drawing that looks correct. The order of each individual segment is drawn sometimes in an unnatural way if you think in terms of plotting or machining. Deciding two lines that seem to end at a nearby location should be connected or not is risky business. What can we do, however is consider every segment as a different operation and allow the user to group them together lately. This way we remove most of the dangerous decisions from the code and leave it to the user instead.

VernonEnslin, 2015-06-15:
Hi Peter nice to see you!  Me and my friends were just wondering about those OPEN HARDWARE open source watches you promised us that we already payed for last year April, Thank you tbfleming for al your hard work we really appreciate  it!

tbfleming, 2015-06-15:
@anthonywebb, @misan is correct; dxf originated in an era when the about the only thing CAD was used for was for printing. Commercial CADs which are designed for manufacturing abandoned DXF-style files and switched to solid model formats.

openhardwarecoza, 2015-06-15:
@anthonywebb I can give you a collection! I have several customers ordering metal lasercutting on a weekly basis.  I will zip up a few of the 'open' designs - which tend to be worst case (:

Imagine person-X taking a i3 frame DXF and adding their own logo (:

But yeah I do have a good collection for you to play with.

openhardwarecoza, 2015-06-15:
@VernonEnslin  - wrong platform but See the mail I sent to your vernone927 address on 17 Aug 2014:

Peter van der Walt   8/17/14

to Dennis, Corrie, vernone927 
Sending off Corrie's, Dennis and Vernon to House4Hack, 4 Burger Ave via Fastway courier tracking no NA0003728652 tomorrow.  I addressed it for Att Dennis - refering a whatsapp conversation on Friday

Check with Dennis - parcel was addressed to him.  You could also call Fastway and get them to pull the waybill for a signature... that ones done on my books...

Akfreak, 2015-06-16:
Just curious is DXF just yiur preferred file format, or os DXF the standard for saving artwork for gcode conversion.

openhardwarecoza, 2015-06-17:
Dxf is pretty much the go to standard for people sending mechanical parts to companies for laser/cnc/waterjet work.  

My preferrrd format is scad,  or sketchup lol.  Its not personal,  its industry wide. 

I came across https://github.com/xenovacivus/PathCAM a few mins ago but wont be near a pc till day after tomorrow (browing on my phone)  and i noticed pathcam uses STL.  @tbfleming if dxf sucks so bad,  how fo you feel about STL?  No reason a 2d drawing cant be a stl.  Heck 3d stl might make for easy machining setup !  

Plus i have a easy dxf - stl workflow since i often have okes being used to the old way bringing in 2d dxf,  and wants me to 3d print it.  (:

openhardwarecoza, 2015-06-17:
@misan now that i read your post again,  in cambam that was how it was handled too. Select  a bunch of independent segments,  hit ctrl+j to Join.    That would also be a totally acceptacle workflow step in jscut

openhardwarecoza, 2015-06-17:
@misan 's plan would work even better uf you combine it with https://github.com/tbfleming/jscut/issues/19  -  prior to export,  or in qcad you could color objects that should be one path,  alike.   Then use the select panel to select all the for example,  red lines, click join and voila

openhardwarecoza, 2015-06-17:
Excuse all the typos.  Typing off phone tonight.  And githubs mobile interface doesn't have Edit :(

tbfleming, 2015-06-17:
STL files which aren't broken are easier to handle than DXF. The two problems with STL files are:
* Broken files
* They are 3d; jscut is 2d. (fixable for prismatic parts, assuming they are oriented Z-up; potentially difficult to explain this to new users)

tbfleming, 2015-06-21:
@openhardwarecoza, please stop complaining every place jscut is mentioned and just find someone who will meet the requirements I set out above:

> If someone is willing to write a repair tool that works well, and is willing to relieve me of having to answer support questions for DXF files that have problems after conversion, then I'll be willing to integrate it.

closed 2015-06-25 by openhardwarecoza