tbfleming/jscut #45: Larger cut that expected in a simple square

bensuffolk, 2015-06-17:
First let me say I’m new to CAM and CNC so I may well have hold of the wrong end of the stick on what the different operations like inside, outside etc do. But I really like the concept of jscut, you have done a great job so far on it! Its great to have something to use from OS/X :-)

I’m using a 3.175mm tool and to try and figure out what happens I made a very simple filled square in inkscape, is 10mm x 10mm and looks like this :

    <rect
       style="fill:#000000;fill-opacity:1;stroke:none;stroke-width:3.25084114;stroke-miterlimit:3.4000001;stroke-dasharray:none;stroke-opacity:1"
       id="rect4710"
       width="35.433071"
       height="35.433071"
       x="3.5433071"
       y="1013.3858" />

So I know inkscape is 90 dots per inch (DPI), that means its (90/25.4) dots per mm (DPMM). I can see the width and height is 35.433071 mm. Dividing that by the DPMM the width and height are 10.000000038 mm. So thats as I expect.

I pop that into jscut and ask it to do an inside and this is the relevant bit of GCode :

; cut
G1 X0.0000 Y0.0000 F200
G1 X0.0000 Y7.4917
G1 X7.4917 Y7.4917
G1 X7.4917 Y0.0000

So if we add the Tool width to that its cut will be from -(3.175/2) to 7.4917 + (3.175/2) which is a total cut size of 10.6667mm. That making the square just over 6% larger than I would have expected it to be. Is this just a pile of rounding errors adding up to generate this, or is there something I have done wrong?

tbfleming, 2015-06-17:
Please use "Save Settings" -> Gist and report the ID back so I can see how you have everything set up.

bensuffolk, 2015-06-17:
https://gist.github.com/anonymous/beb44618cc480f7f40a1

tbfleming, 2015-06-17:
It's missing your SVG & Operation; please save after the point where you generate gcode.

bensuffolk, 2015-06-17:
Sorry, never used gist before. 

https://gist.github.com/anonymous/55e3b649f3ff6b80a892

tbfleming, 2015-06-17:
Here's the easy part: You're using Inkscape 0.91 (new version), which switched to 96 px/in. Use 96 in jscut and it works.

Here's the hard part: Inkscape does some funny business inside the headers which make the inside of the file look correct to people when read as 90 px/in. To jscut and other apps though, it's 96 px/in.

bensuffolk, 2015-06-17:
Ah ok, thanks for pointing that out, how very odd that it all looked right in the file. Now I know I'll set it to 96 and all will be good :-)

closed 2016-05-16 by tbfleming