tbfleming/jscut #46: Rapid feedrate too high on Z axis

kelna2, 2015-07-18:
I am using a Shapeoko 2 with a Due and a gShield running TinyG2. My Z axis is controlled using an ACME threaded rod, where as my XY are controlled by belts and pulleys.

My Z axis on my CNC cannot move at the same rapid speed as my XY axes can. It would be nice if there was an option in jscut to set a Z axis Rapid speed that was separate from the XY Rapid speed.

I have gotten around this issue by setting the rapid speed to a low value, but it's really time consuming to watch the CNC crawl during an XY movement.

Thanks!

tbfleming, 2015-07-18:
Did you try TinyG's maximum Z feedrate setting?

kelna2, 2015-07-19:
Oh good call! haha :)

It's kind of a pain to adjust settings on a TinyG2 since it doesn't have eeprom to store the settings you have to edit the souce code, re-compile, and upload it. I'm re-compiling now for my machine, but i'm sure others would benefit from having the option to set their rapid speed on Z.

Thanks for the help!

closed 2015-07-19 by kelna2