tbfleming/jscut #48: make failed

framlin, 2015-11-24:
I wanted to make a standalone local version of jscut, but it allways fails with the following message:

framlin@bach:~/src/jscut$ make default
cd cpp && em++ cam.cpp hspocket.cpp separateTabs.cpp vEngrave.cpp -I /home/framlin/src/jscut/lib/boost_1_56_0 -std=c++11 --memory-init-file 0 -fcolor-diagnostics -Wall -Wextra -Wno-unused-function -Wno-unused-parameter -Wno-unused-variable -Wno-logical-op-parentheses -s ASSERTIONS=0 -s ALLOW_MEMORY_GROWTH=1 -s SAFE_HEAP=0 -s DISABLE_EXCEPTION_CATCHING=1 -s FORCE_ALIGNED_MEMORY=1 -s NO_EXIT_RUNTIME=1 -s EXPORTED_FUNCTIONS="['_hspocket', '_separateTabs', '_vPocket']" -o ../js/cam-cpp.js  -O3 --llvm-lto 1 

[...] some warnings [...]

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 389, in _handle_results
    task = get()
TypeError: ('__init__() takes at least 3 arguments (1 given)', <class 'subprocess.CalledProcessError'>, ())

any hints, what I could do?

tbfleming, 2015-11-24:
Initial guess: your Python version and emscripten versions aren't compatible with each other. Can you compile other things using emscripten?

framlin, 2015-11-24:
hm, I have not tried that. I have just installed emscripten and called make ...

framlin, 2015-11-24:
no, I cannot. 

I tried a hello_world.c

/usr/bin/lli: error creating EE: No available targets are compatible with this triple, see -version for the available targets.

it seems, I have to learn how emscripten works, first :-(

framlin, 2015-11-24:
meanwhile I have made a new installation of emscripten, that was able to complie a hello_world.c
But jscut is still not able to be made with the provided Makefile:
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++
framlin@bach:~/src/jscut$ make
cd cpp && em++ cam.cpp hspocket.cpp separateTabs.cpp vEngrave.cpp -I /home/framlin/src/jscut/lib/boost_1_56_0 -std=c++11 --memory-init-file 0 -fcolor-diagnostics -Wall -Wextra -Wno-unused-function -Wno-unused-parameter -Wno-unused-variable -Wno-logical-op-parentheses -s ASSERTIONS=0 -s ALLOW_MEMORY_GROWTH=1 -s SAFE_HEAP=0 -s DISABLE_EXCEPTION_CATCHING=1 -s FORCE_ALIGNED_MEMORY=1 -s NO_EXIT_RUNTIME=1 -s EXPORTED_FUNCTIONS="['_hspocket', '_separateTabs', '_vPocket']" -o ../js/cam-cpp.js  -O3 --llvm-lto 1 
ERROR:root:Compiler settings are incompatible with fastcomp. You can fall back to the older compiler core, although that is not recommended, see http://kripken.github.io/emscripten-site/docs/building_from_source/LLVM-Backend.html
Traceback (most recent call last):
  File "/usr/local/bin/em++", line 12, in <module>
    emcc.run()
  File "/opt/emsdk_portable/emscripten/incoming/emcc.py", line 918, in run
    raise e
AssertionError: forced aligned memory is not supported in fastcomp
Makefile:63: recipe for target 'default' failed
make: *** [default] Error 1

+++++++++++++++++++++++++++++++++++++++++++++++++

Aftger I have removed the 
    -s FORCE_ALIGNED_MEMORY=1                       \

it seems to compile.

Do you think there could be some trouble because of the removed FORCE_ALIGNED_MEMORY-Flag?

tbfleming, 2015-11-24:
FORCE_ALIGNED_MEMORY=1 is an aggressive optimizer setting; it's safe to
remove.

g19fanatic, 2016-04-11:
Since gh-pages is serving jscut.org, shouldn't I be able to just git clone that branch to have an offline version of jscut?

tbfleming, 2016-04-11:
The Makefile produces cam-cpp.js. That file's too big for gs-pages, so it's not in the repository. Instead, the online version fetches it from api.jscut.org, which is served by Google App Engine.

Another difference: gs-pages and standalone use a different config.js. To fix: copy config_standalone.js over config.js.

tbfleming, 2016-04-12:
You can download cam-cpp.js from http://api.jscut.org/js/cam-cpp.js . Place it in the js directory.

closed 2016-05-16 by tbfleming