tbfleming/jscut #49: Use with LinuxCNC

tiger12506, 2015-12-12:
I tried to get jscut to play nicely with LinuxCNC a while back, but I didn't know much g-code at the time, and the error message from LinuxCNC wasn't very helpful.

Now I recognize that jscut is leaving off M30 at the end of the file. Adding it manually works fine.

Could this be added by default or as an option?

ebpa, 2016-05-13:
Perhaps the UI should have preamble / conclusion sections for manually entering g-code and/or elements for common commands (spindle/coolant control, etc.).  "Return to 0,0 at end" would have a good home.

tbfleming, 2016-05-16:
M30 is "exchange pallet shuttles and end the program"; I doubt anyone with a $1M machine that does that would risk using jscut.  It now adds M2 to the end of file.

tiger12506, 2016-07-07:
Sorry I haven't addressed this sooner, my machine was down for repairs.
The M2 at the end, tested, works for me. Thanks! From my perspective as OP, it's okay to close this issue.

closed 2016-07-07 by tbfleming