tbfleming/jscut #50: Path header and footer

serealhansman, 2016-03-16:
Hi!
I am currently putting together a Plasma Cutter CNC Machine.
The Controller is an Arduino running GRBL, using grblweb in combination with jscut as a Frontend.
For the Plasma Cutter to work, i Need some Special G-Codes at the beginning of each Path (Probing the Surface, starting the Plasma, moving up a bit, waiting some time, moving a Little down) and also at the end (stopping Plasma, moving a Little up, waiting some time).

The easiest way to achieve this - i think - would be to include a headerfile to the GCode-Output before each path, followed by a footerfile after each path...

What Do you think about that? Could this be possible?

danielkucera, 2020-05-08:
I would also need this, I'd like to use it for laser cutting and instead of plunge and retract I need to run gcode to start/stop laser.
I would be nice to have two text fields where one can put gcode which would be used instead of this:
https://github.com/tbfleming/jscut/blob/88652b7bc9b19a854cc39b49b5bbed98d0c473d5/js/Cam.js#L374
and this:
https://github.com/tbfleming/jscut/blob/88652b7bc9b19a854cc39b49b5bbed98d0c473d5/js/Cam.js#L473

tbfleming, 2020-05-08:
LaserWeb4 is a better match for laser cutting. It's the successor to jscut.

danielkucera, 2020-05-08:
I have just found it. You should add it to the readme :)