tbfleming/jscut #51: Tabs and Retract to clearance

nsiatras, 2016-08-05:
Hello,

Everytime I add tabs to the design I want to cut the generated gcode keeps sending the Z axis to the clearance level before moving it to the next pass depth.

Is there any way to change that ?

tbfleming, 2016-08-06:
It normally doesn't do that. Can you post a test case? Save Settings -> Gist then post the link here.

nsiatras, 2016-08-06:
Here are the settings but I dont think it has to do with that
https://gist.github.com/anonymous/a321e9139d5f67a075ad23ff09197220

tbfleming, 2016-08-06:
You saved before you loaded an SVG and added your operations and tabs. If you save after you do all that I'll get a complete reproduction. Make sure you're OK sharing your SVG before doing this.

nsiatras, 2016-08-06:
Sorry. I saved it again with an svg loaded.
https://gist.github.com/anonymous/43ce313738bf094d80e7e93288932f8a

tbfleming, 2016-08-06:
Very odd. That issue appears to be tied to the shape.

nsiatras, 2016-08-06:
The same bug happens with all shapes. Even a simple circle or square.
Try make a simple square, add tabs and you will see your own.