tbfleming/jscut #53: Ability to select Circles and Lines and use line thickness as filled path

brantwedel, 2017-03-14:
Need a way to select lines with thickness, and shapes from svg files.

I came across this when exporting from Fritzing, it uses circles and lines to represent pads and traces on pcbs.

Found some resources and made a prototype that converts all shapes/lines to filled paths so I could convert my pcb designs, will try to make a PR after some cleanup:

To "flatten" svg shapes to paths:
https://gist.github.com/timo22345/9413158

To expand unclosed stroked paths/lines to filled paths:
http://microsoft.github.io/maker.js/docs/advanced-drawing/#Outlining%20a%20model

tbfleming, 2017-03-14:
Use Inkscape to transform your SVG. It's loaded with helpful things such as Object to Path and Stroke to Path.

brantwedel, 2017-03-14:
Yep, started with that, for complex pcb designs it is a bit of a hassle, luckily my prototype works well enough for my purposes, thanks.

closed 2017-03-14 by brantwedel