tbfleming/jscut #55: Simplified g-code view for those unlucky enough to not have WebGL working

Harvie, 2017-04-22:
I can't get WebGL working in chrome because of some problems with my drivers. Can you please make some simplified version of g-code simulator that would show at least some static 3D views from various angles?

tbfleming, 2017-04-22:
WebGL is the best way to produce 3D views in the browser; other methods would take a large amount of development time and benefit few users.

closed 2017-04-22 by tbfleming

Harvie, 2017-04-22:
You can easily produce 2D output. Eg. using canvas. Just ignore one axis and voila. You have 2D image of 3D code. Do this three times, once for each axis to be ignored and you have 2D view from 3 different angles. Quite easy.

tbfleming, 2017-04-22:
No, it requires a completely different code base. See https://www.youtube.com/watch?v=tgVLb6fOVVc for an intro to WebGL; it's nothing like the 2D canvas.

tbfleming, 2017-04-22:
You're going to run into more and more pages that rely on WebGL, even for 2D stuff. WebGL outperforms the 2D canvas when you need to draw roughly 10,000 lines or more, assuming you use the WebGL API well. WebGL scales to around a million lines at a decent frame rate, something that causes the 2D canvas to choke.