tbfleming/jscut #56: Show me dimensions of everything

Harvie, 2017-04-22:
I have imported SVG and then i set px per inch. That should be enough to show me how big will the picture be in real life... Maybe have some tool to check distance between two points in SVG... And after generating G-code we should also be able to compute total dimensions of it.

Something like ugs-platform does. It's super useful to check if dimensions are about sane in code:

[image]

Right now i have to export g-code, check dimensions in ugs and then do another export if i need to fix something... I also think it should separately count negative and positive Z dimensions. Because stuff under Z0 is usualy what is actually being cut, while stuff over Z0 is just safety margin and moving over the stock. BTW this should also work when WebGL support is not available...

tbfleming, 2017-04-22:
LaserWeb4 is jscut's successor. It supports this, but requires WebGL to function.

http://cncpro.co/

closed 2017-04-22 by tbfleming

Harvie, 2017-04-22:
LaserWeb4 is for laser cutters. Doesn't seem to be suitable for my milling machine...

tbfleming, 2017-04-22:
LW4 is for a variety of machines; its name is historical from when it used to only support lasers. Look through the postings at https://plus.google.com/communities/115879488566665599508 ; you'll see examples where people used it for lasers, CNC routers, and some esoteric things like a motorized etch-a-sketch.

To enable milling machine support, turn on Settings -> Application -> Enable CNC Mode.

Harvie, 2017-05-23:
So should i take jscut as officialy dead project? I am not really happy with LW4, and i prefer jscut. But i don't see the point why you're still accepting new issues, if you are not willing to continue with development...

nsiatras, 2017-05-23:
I have to agree. I also prefer JSCut to LW. It is a better software for easy CNC operations.

Harvie, 2017-05-23:
@nsiatras hey baby! wanna fork? :)

nsiatras, 2017-05-23:
Wish I could. I am running a few heavy weight projects at the moment so no time for the community.