tbfleming/jscut #65: Cut options for laser

MrSage, 2019-12-21:
Is it possible to have settings for the cutting path? I'm working with a laser on my CNC and this project is the best I've found so far outside of using an older version of Inkscape and some plugins. 

Ideally with laser cuts it could be specified to make all paths linear in the same direction, that way the engraving is even. I'm willing to help out with this if you could give me some pointers.

tbfleming, 2019-12-21:
LaserWeb4 is designed for lasers and is based on an enhanced version of jscut's engine.

MrSage, 2019-12-21:
Oh, thank you! It's really hard to find software in this space so I appreciate the knowledge share. LaserWeb4 spikes my CPU pretty hard but I can at least generate the gcode I need.

closed 2019-12-21 by MrSage