tbfleming/jscut #71: Select all / rectangle missing

WolfgangFahl, 2021-01-27:
Great tool. I'd love to use it. My svg doesn't seem to be a single object - i created it with the convert command of imagemagick.

IMHO it would be much more userfriendly to select all object in a given frame or using a lasso selection as is possible with the mac preview.

WolfgangFahl, 2021-01-27:
see https://gist.github.com/brantwedel/d028c1bdf1acb34bb00d702913d948b3?

WolfgangFahl, 2021-01-27:
see https://groups.google.com/g/jscut/c/64o7__tjGNY?pli=1

WolfgangFahl, 2021-01-27:
"_so the answer is "no, you cannot select all in jscut, but you can preprocess your file in another software to join the pieces"
this seems odd to me. In fact, it seems odd to me that you have to select anything in your svg file- shouldn't the default be all selected?_" ... exactly!

Yardie-, 2022-10-29:
I found a way to do this easily
In inkscape after turning the objects of say a PCB into paths 
select them all 
then path > combine (Ctl + K)
save the svg then import it into jscut and click on one component. and voila they are all selected.
This is perfect for PCB stencils. 
Export your paste mask then do the above in inkscape.