tbfleming/jscut #81: Engrave mode closes paths

prostolyubo, 2023-05-12:
When I select a curve that is not a closed path, jscut will close a path by connecting its two ends with a straight line and engrave this line (it should not be there).
This should either be fixed or turned into a feature by adding a toggle: auto-close paths.

tbfleming, 2023-05-12:
jscut's core algorithms are polygon manipulation algorithms; they can't handle open paths.

prostolyubo, 2023-05-13:
If you have enough time you can code anything. Right now I had small enough project that I could manually edit the generated GCode and get rid of the additional lines. If I'll ever have big enough project that it will be worth making this change then I'll fork and try. Maybe it will just come with certain limitations - like single path per svg object.